wifidog/wifidog-gateway #251: authenticate guest one time

staurasi, 2016-04-08:
n4n0B0y asked for this in the auth-server section:
https://github.com/wifidog/wifidog-auth/issues/10

I ask it here since I think it first depends on the gateway.
This would be a great courtesy feature. Many other captive portal solution provide this nowadays.
Some smartphones have options for remember login credentials, but they are often hidden and most people don't want to mess up with options.

At the moment I emulate this behavior setting a very high timeout in wifidog.conf (12 or 24 hours) and consequently a longer time for DHCP lease time and auth server garbage collecting operations. This of course is not the way and it's tolerable only for a small number of clients. Plus it makes the statistics about login sessions quite unusable.

Here is my opinion on how to accomplish that. The gateway should pass the client MAC address to the auth server as soon as possible. The auth server will search for that MAC in the connection table, choose a relative valid user and if it finds it, login the client automatically. But we cannot do this for all login request. We have a lot of login page request from smartphone apps of unauthenticated users and we don't want the auth server to query the connection table for all those requests.
Maybe a list of MACaddress and logout_timestamp kept by the gateway can solve this. Records in this list can be kept for a specified amount of days (# of days in wifidog.conf and disabled by default for those who need more secure environment).

At a first glance the whole thing would act like:
client asks gateway to pass
gateway check the "logged out recently list"
 if in list pass the MAC to the server
 else proceed as normal
the auth-server receive the request
if MAC is not a parameter redirect to login page
if MAC is a parameter
  search for a valid user
  if found login the client automatically
else redirect to login page

Let me know what you think about it. Do you think it would be useful for the community?
In that case I hope someone will work on that. Otherwise I can try to contribute even if I'm a PHP and C/C++ beginner.

gareth41, 2016-04-09:
Implementing DHCP into wifidog as I explained in wifidog/wifidog-auth#10 is probably a better option, because then you can catch the mac address using dhcp before the client actually gets an ip address.  You could have each client added to the client list when a dhcp discover packet is received, and wifidog tries to authenticate the client mac with the auth server before returning DHCP ACK, this way you won't have problems with iphones etc... trying to find a captive portal before wifidog has had a chance to authenticate the mac against the auth server.  You don;t need to worry about multiple auth requests with mac address becuase it should only happen once on DHCP request, then won't happen again until the client is timed out and removed from the client list.

n4n0B0y, 2016-05-06:
@gareth41  implementing dhcp in the GW is something hard to do and might take time, we should think to an easy and smart tricks to do so.

"HTTP.c" is responsible for getting the MAC @ and redirecting the guest to login page, the basic idea is the get the MAC (already available in /tmp/dhcp.lease) and POST the couple {MAC, GWID} to auth server to check if the user has already connected or not (a simple lookup in connections table) if it returns TRUE means the guest already pass by the gateway we should redirect him to Welcome page (with simple connect button) else (returns FALSE) we proceed as normale by redirecting to login page.

What do you think of this?

n4n0B0y, 2016-05-06:
an other suggestion is to do it under auth.c by adding an other Authentication server response, something like KNOWEN_USER (code:2) then redirect to portal

Tr4cer0ute, 2016-12-16:
@staurasi I have managed to do this from Auth Server side. If you need more information I'am open to share.