wifidog/wifidog-gateway #255: Allow whitelisting (zero-rating) a hostname (IP address)

nathansamson, 2016-05-09:
Hi,

I would like to be able to effectively zero-rate traffic to a given IP address (range). Ideally this would be a per-connection setting (some users get the zero-rate, others don't)

I don't think at the moment something like this is possible (certainly not per user as firewall rules seems pretty static).

I can give it a try developing this myself but I need a few pointers. I think the bulk of the logic is going to be in fw_iptables.c.
From what I can see I need to add a new route-set that would bypass WiFiDog_Outgoing/WiFiDog_Incoming for the given src/dest IP's.

JJEU, 2016-07-12:
you can do that easily : 
create a DHCP reservation for your user
in wifidog.conf, add the mac adress of your user in TrustedMacList

nathansamson, 2016-07-12:
You are misunderstanding the request.
What you are describing is allowing a user to use the system for free.

What I want to achieve is make acess to a given IP(range) free. Like 0.facebook.com is sometimes free on some 3G carriers.

But this should only be free for some connected users; not all of them

JJEU, 2016-07-12:
ok so you want to authorize a domain, but with an ip_source condition

something like this : 
FirewallRule allow tcp to 0.facebook.com from ipsource

where ipsource is the ip or iprangeof your allowed clients. you can use a custom mask to allow only a part of your total ip range, then do a DHCP reservation for your clients on ips on this range
ie 192.168.1.0/xxx

problem is, i don't know if the "from" switch is supported. maybe you can test

alternatively you can create a script witch will create the iptables rules. something inspired by this : #14
see the script posted by missionrulz, gives an exemple.