wifidog/wifidog-gateway #258: Wifidog optimization for very high usage case uses

JJEU, 2016-07-12:
Hello it's Jackson from Ile Sans Fil

We take care of the public wifi in some crowded places in Montreal (ie, wifi areas for the CHUM, and a few big public librairies)

The DHCP server sometimes has more than 1000 different leases in use.

In those places I use openwrt x86-64 trunk on a APU2 hardware from PC engines (x86-64 mini PC, 2ghz quad core CPU, 2/4GB RAM)
all 4 cores shows on openwrt and 4GB ram too.

Sometimes i log remotely on a node, and see the CPU usage a 1.50 and more. most of the CPU is used by wifidog. Users complains that the portal page is very slow to appear.

By default, is wifidog optimized to run on multi-core hardware ? do i have to use a switch to enable it ?

I know it's difficult to reproduce such an extreme case-use, so I could give remote access if you want to investigate.

oleg-umnik, 2016-07-19:
You should check the rate with which connections redirected to wifidog HTTP server. Scan your log file for strings like "Received connection from <IP>, spawning worker thread ". I saw some broken HTTP clients starting HTTP requests flood when get something unexpected from server (i.e. redirect to portal when user is unauthenticated).

liudf0716, 2017-01-26:
i think u can try apfree wifidog, which were used widely in china's business project.
this is project's address
https://github.com/liudf0716/apfree_wifidog

Tr4cer0ute, 2017-01-26:
@liudf0716 can not see how it's different from Wifidog projet. Please take a momebt to giveout 3 benefits of it.

liudf0716, 2017-01-26:
Improved performance
improved stability
More function
In addition, it used widely in real business project and inspired by real user demand

liudf0716, 2017-01-26:
In fact, u can easily find apfree wifidog project is active, however this wifidog project not update for a long time

Tr4cer0ute, 2017-01-26:
@liudf0716 maybe but on apfree config file I don't see support for domain name management. How are u handling this? for example to allow/block specific websites

liudf0716, 2017-01-27:
Of course we support allowing specific website,  I think no need to block website, we can allow it without restart wifidog through wdctl command. If u read code closely, u will find I extend wdctl many function, which all come from real users’ needs

Tr4cer0ute, 2017-01-27:
@liudf0716 I'm talking about Walled garden feature without using WDCTL. Point me to that line of code

liudf0716, 2017-01-27:
If u read wdctl

liudf0716, 2017-01-27:
U can read conf.c, all config file related operation can be found there. all wdctl related function can find their related implementation in config file.

liudf0716, 2017-01-27:
I am now in holiday, so not convenience to paste the exact line code here, sorry for that

Tr4cer0ute, 2017-01-27:
@liudf0716 I see sorry for inconvenience ;-) I'm compiling it to give it a try in production then I will revert with my feedback.

liudf0716, 2017-01-28:
thanks for your question and recommendations .:-)

KazukiShimada, 2017-02-06:
@liudf0716 
Hi. I had a look to your apfree_wifidog.
Awsome work!

It seems you implemented thread pool which lead to perfomance improvement. Right?

liudf0716, 2017-02-07:
@KazukiShimada yes, I useed thread pool to improve performance, It had some effects, but not that good as i had thought.

qianguozheng, 2017-02-07:
@liudf0716  glad to see you here.

liudf0716, 2017-02-07:
@qianguozheng glad to see u too:D. the qq group is ok, I don't known why u find it not active.
In fact, the reason I imported libevent was to resolve https redirecting feature, only libevent provide open source solution. in my todo list, I will replace libhttpd with libevent step by step, for our project was deployed in business scenarios, we must be cautious.

liudf0716, 2017-02-07:
@qianguozheng my qq is 407652474