wifidog/wifidog-gateway #263: src/gateway.c line 370

fbichina, 2016-09-26:
Line 370 in src/gateway.c
"if ((!config) && (!config->pidfile))"
Isn't it wired to write this ???

qianguozheng, 2016-09-30:
something wrong ?

fbichina, 2016-09-30:
`if ((!config) && (!config->pidfile))`
equals
`if ((config==NULL) && (config->pidfile == NULL))`
if "config" is NULL , how to get "pidfile" from NULL??

qianguozheng, 2016-10-11:
if config is NULL then !config is true. it will not get the pidfile

fbichina, 2016-10-11:
@qianguozheng 
Yep, if config is NULL, which means !config is true and the left part of "&&" operation is true, it will go to see whether the right part is also true. That means when config is NULL, it will excute (!config->pidfile)...But how to get a thing from NULL ???

qianguozheng, 2016-10-11:
Indeed, it MUST be wrong

poolborges, 2016-10-20:
That condition is not valid when only pid is null. 
So it must have a or between righ side and left side
We read: when conf is null ou pid is null do something.
if ((!config) || (!config->pidfile))

The code will be better to read if is write li this

if ((config == NULL) || (config->pidfile || NUL))

poolborges, 2016-10-20:
if ((config == NULL) || (config->pidfile == NULL))

fbichina, 2016-10-20:
@qianguozheng NONONONO，，，，It should be like "if( config && config->pid )" cause the next line is to use the config->pid....So the "config" and "config->pid" both can not be null.
For more pretty, it could be written like "if (config != NULL && config->pid != NULL) "  Ha.

qianguozheng, 2016-10-20:
So this issue can be closed

closed 2016-10-20 by fbichina

poolborges, 2016-10-20:
Depende of whow important is to save pid

**The Original is**
``
/* save the pid file if needed */
if ((!config) && (!config->pidfile))
    save_pid_file(config->pidfile);
``
**The Suggest**
if ((config == NULL) || (config->pidfile == NULL)){
    debug(LOG_ERR, "Could not get PID, exiting...");
    exit(1);
}else{
    save_pid_file(config->pidfile);
}