wifidog/wifidog-gateway #264: too many redirects occurred when open Portal use ios

chijiao, 2016-09-27:
log like this

[7][Tue Sep 27 09:39:13 2016][5994](ping_thread.c:232) Auth Server Says: Pong
[6][Tue Sep 27 09:39:48 2016][5994](gateway.c:474) Received connection from 192.168.99.240, spawning worker thread
[7][Tue Sep 27 09:39:48 2016][5994](httpd_thread.c:65) Processing request from 192.168.99.240
[7][Tue Sep 27 09:39:48 2016][5994](httpd_thread.c:66) Calling httpdProcessRequest() for 192.168.99.240
[6][Tue Sep 27 09:39:48 2016][5994](http.c:124) Got client MAC address for ip 192.168.99.240: 1c:5c:f2:b8:85:90
[6][Tue Sep 27 09:39:48 2016][5994](http.c:134) Captured 192.168.99.240 requesting [http%3A//captive.apple.com/hotspot-detect.html] and re-directing them to login page
[7][Tue Sep 27 09:39:48 2016][5994](http.c:214) Redirecting client browser to http://wifi.hideus.net:80/wx/?gw_address=192.168.99.1&gw_port=2060&gw_id=default&mac=1c:5c:f2:b8:85:90&url=http%3A//captive.apple.com/hotspot-detect.html
[7][Tue Sep 27 09:39:48 2016][5994](httpd_thread.c:68) Returned from httpdProcessRequest() for 192.168.99.240
[7][Tue Sep 27 09:39:48 2016][5994](httpd_thread.c:73) Closing connection with 192.168.99.240
[6][Tue Sep 27 09:39:49 2016][5994](gateway.c:474) Received connection from 192.168.99.240, spawning worker thread
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:65) Processing request from 192.168.99.240
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:66) Calling httpdProcessRequest() for 192.168.99.240
[6][Tue Sep 27 09:39:49 2016][5994](http.c:124) Got client MAC address for ip 192.168.99.240: 1c:5c:f2:b8:85:90
[6][Tue Sep 27 09:39:49 2016][5994](http.c:134) Captured 192.168.99.240 requesting [http%3A//www.apple.comhttp%3A/www.apple.com/] and re-directing them to login page
[7][Tue Sep 27 09:39:49 2016][5994](http.c:214) Redirecting client browser to http://wifi.hideus.net:80/wx/?gw_address=192.168.99.1&gw_port=2060&gw_id=default&mac=1c:5c:f2:b8:85:90&url=http%3A//www.apple.comhttp%3A/www.apple.com/
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:68) Returned from httpdProcessRequest() for 192.168.99.240
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:73) Closing connection with 192.168.99.240
[6][Tue Sep 27 09:39:49 2016][5994](gateway.c:474) Received connection from 192.168.99.240, spawning worker thread
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:65) Processing request from 192.168.99.240
[7][Tue Sep 27 09:39:49 2016][5994](httpd_thread.c:66) Calling httpdProcessRequest() for 192.168.99.240
[6][Tue Sep 27 09:39:49 2016][5994](http.c:124) Got client MAC address for ip 192.168.99.240: 1c:5c:f2:b8:85:90
[6][Tue Sep 27 09:39:49 2016][5994](http.c:134) Captured 192.168.99.240 requesting [http%3A//wifi.hideus.nethttp%3A/wifi.hideus.net/wx/%3Fgw_address%3D192.168.99.1%26gw_port%3D2060%26gw_id%3Ddefault%26mac%3D1c%3A5c%3Af2%3Ab8%3A85%3A90%26url%3Dhttp%253A/www.apple.comhttp%253A/www.apple.com/] and re-directing them to login page

mhaas, 2016-09-27:
Can you attach your wifidog.conf?

chijiao, 2016-09-28:
@mhaas 

# $Id$
# WiFiDog Configuration file

# Parameter: GatewayID
# Default: default
# Optional
#
# Set this to the node ID on the auth server
# This is used to give a customized login page to the clients and for
# monitoring/statistics purpose. If you run multiple gateways on the same
# machine each gateway needs to have a different gateway id.
# If none is supplied, the mac address of the GatewayInterface interface will be used,
# without the : separators

GatewayID {default}

# Parameter: ExternalInterface
# Default: NONE
# Optional
#
# Set this to the external interface (the one going out to the Inernet or your larger LAN).  
# Typically vlan1 for OpenWrt, and eth0 or ppp0 otherwise,
# Normally autodetected

ExternalInterface eth0

# Parameter: GatewayInterface
# Default: NONE
# Mandatory
#
# Set this to the internal interface (typically your wifi interface).    
# Typically br-lan for Openwrt (by default the wifi interface is bridged with wired lan in openwrt)
# and eth1, wlan0, ath0, etc. otherwise
# You can get this interface with the ifconfig command and finding your wifi interface

GatewayInterface wlan0

# Parameter: GatewayAddress
# Default: Find it from GatewayInterface
# Optional
#
# Set this to the internal IP address of the gateway.  Not normally required.

# GatewayAddress 192.168.1.1

# Parameter: HtmlMessageFile
# Default: wifidog-msg.html
# Optional
#
# This allows you to specify a custome HTML file which will be used for
# system errors by the gateway. Any $title, $message and $node variables
# used inside the file will be replaced.
#
# HtmlMessageFile /opt/wifidog/etc/wifidog-.html

# Parameter: AuthServer
# Default: NONE
# Mandatory, repeatable
#
# This allows you to configure your auth server(s).  Each one will be tried in order, untill one responds.
# Set this to the hostname or IP of your auth server(s), the path where
# WiFiDog-auth resides in and the port it listens on.
#AuthServer {
#	Hostname    120.76.131.120             (Mandatory; Default: NONE)
#	SSLAvailable   no          (Optional; Default: no; Possible values: yes, no)
#
#	SSLPort                  (Optional; Default: 443)
#	HTTPPort  80               (Optional; Default: 80)
[... 167 lines cut ...]
# Rule Set: known-users
#
# Used for normal validated users.
FirewallRuleSet known-users {
    FirewallRule allow to 0.0.0.0/0
}

# Rule Set: unknown-users
#
# Used for unvalidated users, this is the ruleset that gets redirected.
#
# XXX The redirect code adds the Default DROP clause.
FirewallRuleSet unknown-users {
    FirewallRule allow udp port 53
    FirewallRule allow tcp port 53
    FirewallRule allow udp port 67
    FirewallRule allow tcp port 67
}

# Rule Set: locked-users
#
# Not currently used
FirewallRuleSet locked-users {
    FirewallRule block to 0.0.0.0/0
}

chijiao, 2016-09-28:
GatewayID {default}
ExternalInterface eth0
GatewayInterface br0
GatewayAddress 192.168.20.1
AuthServer {
    Hostname wifi.hideus.net
    SSLAvailable no
    Path /
    PingScriptPathFragment ping/?
    LoginScriptPathFragment /wx/index.php?
    AuthScriptPathFragment /auth/?
    PortalScriptPathFragment /succ.php?
}
HTTPDMaxConn 1
HTTPDUserName admin
HTTPDPassword secret
CheckInterval 60
ClientTimeout 5
FirewallRuleSet global {

}
FirewallRuleSet validating-users {
    FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet known-users {
    FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet unknown-users {
    FirewallRule allow udp port 53
    FirewallRule allow tcp port 53
    FirewallRule allow udp port 67
    FirewallRule allow tcp port 67
}
FirewallRuleSet locked-users {
    FirewallRule block to 0.0.0.0/0
}

fenggui321, 2016-09-28:
@chijiao  you redirct login page "http://wifi.hideus.net:80/wx/?gw_address=192.168.99.1&gw_port=2060&gw_id=default&mac=1c:5c:f2:b8:85:90&url=http%3A//captive.apple.com/hotspot-detect.html" only return "Pong" not a web page.

chijiao, 2016-09-28:
@fenggui321  i know. but man problem not here. i thinks sometimes wifidog parse URL error.  ijust remove url parameter.  working now

```
Index: wifidog-gateway-1.3.0/src/http.c
===================================================================
--- wifidog-gateway-1.3.0.orig/src/http.c
+++ wifidog-gateway-1.3.0/src/http.c
@@ -110,14 +110,14 @@ http_callback_404(httpd * webserver, req
             /* We could not get their MAC address */
             debug(LOG_INFO, "Failed to retrieve MAC address for ip %s, so not putting in the login request",
                   r->clientAddr);
-            safe_asprintf(&urlFragment, "%sgw_address=%s&gw_port=%d&gw_id=%s&ip=%s&url=%s",
+            safe_asprintf(&urlFragment, "%sgw_address=%s&gw_port=%d&gw_id=%s&ip=%s",
                           auth_server->authserv_login_script_path_fragment, config->gw_address, config->gw_port,
-                          config->gw_id, r->clientAddr, url);
+                          config->gw_id, r->clientAddr);
         } else {
             debug(LOG_INFO, "Got client MAC address for ip %s: %s", r->clientAddr, mac);
-            safe_asprintf(&urlFragment, "%sgw_address=%s&gw_port=%d&gw_id=%s&ip=%s&mac=%s&url=%s",
+            safe_asprintf(&urlFragment, "%sgw_address=%s&gw_port=%d&gw_id=%s&ip=%s&mac=%s",
                           auth_server->authserv_login_script_path_fragment,
-                          config->gw_address, config->gw_port, config->gw_id, r->clientAddr, mac, url);
+                          config->gw_address, config->gw_port, config->gw_id, r->clientAddr, mac);
             free(mac);
         }
 

```

OmOiYuMe, 2017-02-18:
Hey which version and architecture are you using? i need to fix this too, would like to use 1.3 for openwrt AR71xx 15.05.1