wifidog/wifidog-gateway #265: best router?

lliiaa, 2016-10-03:
Considering installing wifidog for a coffee shop client; does anyone have best router recommendations beyond the Linksys WRT54G family? If not, which of the WRT54G routers have you used and can recommend?

(Apologies if this information is available somewhere else, but my googling didn't turn up a list!)

t123yh, 2016-10-03:
It's too old. You can use Netgear WNDR4300. Or you can run a OpenWrt-x86.

gareth41, 2016-10-04:
The WRT54G/GL will work, i'm using it here but you need to run the older openwrt kamikaze and somewhat edit the wifidog package makefile so it compiles properly from the github source, I can give you a copy of my makefile if you want, however having said that you're best to get a newer router, something like what t123yh mentioned or even just a cheap tplink will do, they've got more ram and cpu in them, faster wifi and throughput.

Our company has some pre-built firmware images here for different routers with wifidog integrated and pre configured already if you wanted to try the old WRT54G, these images though are setup to work through our system which you're welcome to use free if you want. https://secure.zappie.net/hotspot-firmware.

lliiaa, 2016-10-05:
Thanks for the link, @gareth41! Picked up the TP-Link Archer C7 and will be installing wifidog on that this week, going to try the free Zappie as well. Going to keep this thread open for now in case people want to leave recommendations in the future, or find it via Google and have relevant questions.

Tr4cer0ute, 2016-12-09:
@lliiaa how your experience so far with that router?

turanmete-bhr, 2016-12-09:
Archer C7 nearly best router for wifidog we are using it in over 50 places.

Tr4cer0ute, 2016-12-09:
@turanmete-bhr Thanks.

Tr4cer0ute, 2016-12-10:
What do you think about best criteria to choose a router for WiFidog? Are CPU, RAM and Flash good enough as factors?