wifidog/wifidog-gateway #266: cannot build 1.2.1 and 1.3.0

turanmete-bhr, 2016-10-12:
1) After building 1.3.0 in Caos Calmer when I am trying to run it. I am getting wdctl not found. Like mentioned in [/issues/243](https://github.com/wifidog/wifidog-gateway/issues/243)
2) I cannot even build 1.2.1 getting error in protocol.c while compiling 
`protocol.c: In function '_httpd_catFile':
protocol.c:534:10: warning: implicit declaration of function 'open' [-Wimplicit-function-declaration]
     fd = open(path, O_RDONLY);
          ^
protocol.c:534:21: error: 'O_RDONLY' undeclared (first use in this function)
     fd = open(path, O_RDONLY);
                     ^
protocol.c:534:21: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [protocol.lo] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [/Volumes/OpenWrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/wifidog-normal/wifidog-1.2.1/.built] Error 2
make[2]: Leaving directory `/Volumes/OpenWrt/openwrt/feeds/packages/net/wifidog'
make[1]: *** [package/feeds/packages/wifidog/compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWrt/openwrt'
make: *** [package/wifidog/compile] Error 2
`

LGaitan, 2016-12-12:
Hello @turanmete-bhr ,

I was able to make and install 1.3.0, were you cross-compiling?
Did you ran make? Are the wdctl executable on your /bin or /usr/bin?

kuzjnvn, 2017-02-04:
anybody can make wifidog ipk for 14.07 openwrt ar71xx ? 
thanks so much

kuzjnvn, 2017-02-14:
i built wifidog 1.3.0 for openwrt 15.05 and 14.07 success :)