wifidog/wifidog-gateway #269: iptables child process may not exit normally causing whole gateway to hang

gareth41, 2016-11-07:
This happens when a user authenticates and fw_allow() is called.  On very rare occasions the iptables child process doesn't exit and waitpid() hangs waiting for it.  Whats more the client list lock is also on resulting in the whole gateway being frozen.  execute() needs changing and a timeout implementing, so if a child doesn't exit within x seconds, further action is taken, ie killing the child process, clearing using fw_deny() and then calling fw_allow() again.

oleg-umnik, 2017-01-22:
Could you please try to reproduce described scenario with the following patch?

https://github.com/wifidog/wifidog-gateway/pull/248

liudf0716, 2017-02-06:
I also find such bug, which caused by  execute() function. In fact, not only fw_allow(),  when u start up wifidog, u can also find some iptables rule not execute successly.

liudf0716, 2017-02-06:
I think in multi-thread process, using fork will cause some strange problem

KazukiShimada, 2017-02-06:
@liudf0716 
Do you mean your apfree_wifidog has been fixed it yet?
https://github.com/liudf0716/apfree_wifidog

liudf0716, 2017-02-07:
In fact, apfree wifidog also use execute() heavily in fw operation, anyway I have do my best to avoid using execute(). In the future, I plan to replace iptables's api to replace execute() operation.