wifidog/wifidog-gateway #274: Wifidog Installation on ubuntu/ Debian

oscarmerd, 2017-02-21:
wifidog gateway 1.3.0  doesn't build with ./configure , make and make install

error: configure file not found
error: make and make install file not found

Please assist
[image]
[image]
[image]

mhaas, 2017-02-21:
Please use the most recent release

oscarmerd, 2017-02-21:
Similarly the same error, tried 1.3.0

oscarmerd, 2017-02-21:
Am i missing something on. I did try with original version it works perfectly that is 2009 version, but the latest the command doesn't work.
Please assist.

Merd

lgaitan, 2017-02-21:
if you are cloning directly from github, try executing ./autogen.sh first. It'll generate the autoconf files.

oscarmerd, 2017-02-21:
Still ain't working would this is what i get as the output
[image]

lgaitan, 2017-02-22:
make sure you have libtool and autoconf packages, try apt-get install autotools autoconf automake libtool

oscarmerd, 2017-02-22:
Thank you so much, I installed successfully.

closed 2017-03-05 by florida63