wifidog/wifidog-gateway #281: Keep browser opened after connection is allowed

iffifan, 2017-07-12:
Hello,

We are facing this issue. When a user connects and is authenticated by auth server, browser window is closed automatically on android 5+. We want to keep it open so that we can show some advertisement after user is authenticated.

Any idea?

gareth41, 2017-07-12:
Apple does a similar thing too when it detects that http requests are no longer being redirected, you could try displaying the adds before allowing the client through the firewall.

Tr4cer0ute, 2017-07-12:
@iffifan I think this is related to device manufacturer an how they deal with captive portal.

iffifan, 2017-07-17:
@Tr4cer0ute Can we fix it with firmware? I am ready to pay if anyone can help.

lgaitan, 2017-07-17:
@iffifan You simply have to change how your authenticator works.

The gateway will check with the authenticator if user is already logged in, if so, it will enable futher connections.
Since the device now is able to access the internet, the OS knows the user has logged in and then closes the page.

You just have to change your authenticator behaviour, by only answering the user status after some seconds or after some user action.

iffifan, 2017-07-20:
@lgaitan 
I understand what you mean.
We allow internet after user accepts terms and conditions.
Now if you are saying that we should show ads before allowing internet then it is possible but we also want user to navigate to ad website. So I think it is not about when we allow internet. If browser is closed when internet access is provided then it is an issue. because we want user to have browser opened even after authentication.

gareth41, 2017-07-20:
How about whitelisting the adserver ip/subnet so the add can display before the user is allowed through the firewall.  Display the add to the user with a countdown timer, when the timer has finished internet is allowed.  If the user clicks on the add – internet is allowed from this point onwards and the browser immediately redirected to the advertised website.  This may work even if the user doesn’t see the advertised website before the browser window closes, however because the add was clicked, the browser should still follow the series of redirects until it is redirected no more - resulting in the advertising company registering it as a click.

imnobodyet, 2018-04-17:
@iffifan faced the same problem here, i used a solution to prevent auto portal poping up(it's a system window not a browser), so that users must open a browser to navigate to the portal page, and the browser won't close itself after login.

JJEU, 2018-04-24:
see my comment here https://github.com/wifidog/wifidog-gateway/issues/287