wifidog/wifidog-gateway #282: Not loading CSS for Facebook

chrisbalicki, 2017-09-05:
I am running WiFiDog 1.2.1 on OpenWRT Chaos Calmer 15.05.1.
I am using an external Captive Portal which is working fine, up until when I want to load Facebook in the captive portal (for Facebook Authentication).

When being redirected to Facebook, it seems that the CSS is being blocked and all I can see is the text & hyperlinks. Nonetheless I am still able to login to facebook and authenticate etc, however it just seems the CSS is being blocked.

Does anyone have any ideas what could be causing this and the solution?

I have allowed Facebook via the FirewallRuleSet Global, which looks like this: 
FirewallRule allow tcp to www.facebook.com 
FirewallRule allow tcp to star.c10r.facebook.com 
FirewallRule allow tcp to static.xx.fbcdn.net 
FirewallRule allow tcp to scontent-mad1-1.xx.fbcdn.net 
FirewallRule allow tcp to fbstatic-a.akamaihd.net 
FirewallRule allow tcp to m.facebook.com  
FirewallRule allow tcp to connect.facebook.net 
FirewallRule allow tcp to scontent.xx.fbcdn.net 
FirewallRule allow tcp to graph.facebook.com 
FirewallRule allow tcp to api.facebook.com 
FirewallRule allow tcp to staticxx.facebook.com 
FirewallRule allow tcp to api.twitter.com 
FirewallRule allow tcp to abs-0.twimg.com 
FirewallRule allow tcp to abs.twimg.com 
FirewallRule allow tcp to pbs.twimg.com 
FirewallRule allow tcp to fonts.googleapis.com 
FirewallRule allow tcp to code.jquery.com

gareth41, 2017-09-06:
Use Firefox with the live http headers addin, look for when it tries to load the css files, they are probably sitting on a different server which requires whitelisting through wifidog.

chrisbalicki, 2017-09-06:
Cheers, I updated the list with what I got from Live http headers, still getting the same issue. The issue now seems to be related to the Captive Portal Popup screen, as when I close that and navigate to facebook.com it loads fine. Do you have any suggestions what might be blocking it in the pop out view?

pmosse, 2018-01-30:
Hello! Do you have any updates on this?

mhaas, 2018-01-30:
You cannot whitelist the DNS entries like that: the names are resolved once to a single IP. In reality, they point to multiple IPs in a round-robin fashion.

Look here for details on how this can be implemented on openwrt with dnsmasq: https://github.com/mhaas/fbwlan/blob/master/README.md#allowing-access-to-facebook

closed 2018-01-30 by mhaas

rootcause89, 2018-02-23:
This has actually turned out to be a Apple issue on Macbooks. For whatever reason, it doesn't load the CSS in the Captive Network Assistant. I have found this on all Captive Portal logins (that I had access to) when using a macbook. 

Works fine on all other apple devices, and if you close the CNA and load the splash page in the browser it loads the CSS fine. Bizzarre behaviour, couldn't find any work around, also noticed alot of people have complained about this to Facebook directly in the developers forum but doesn't look like they will work towards a solution.