wifidog/wifidog-gateway #284: HTTPS with Wifidog

chrisbalicki, 2017-10-15:
Is there a straight forward solution to using HTTPS with Wifidog? 

Currently I use HTTPS with my AWS instance which holds the Captive Portal & Auth. The instance forces users to HTTPS and this seems to cause an issue for the Wifidogs heartbeat as it hangs at the 302 error (Please note I have deliberately changed my URL in the following): 

[7][Sun Oct 15 22:32:10 2017][3522](simple_http.c:87) Reading response
[7][Sun Oct 15 22:32:11 2017][3522](simple_http.c:111) Read 640 bytes
[7][Sun Oct 15 22:32:11 2017][3522](simple_http.c:124) HTTP Response from Server: [HTTP/1.1 302 Found
Content-Type: text/html; charset=iso-8859-1
Date: Sun, 15 Oct 2017 22:32:10 GMT
Location: https://www.<MYURL>.com/api/client/heartbeat?gw_id=CLIENT-0001&sys_uptime=1818&sys_memfree=38676&sys_load=0.02&wifidog_uptime=2
Server: Apache
Content-Length: 334
Connection: Close

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="https://www.<MYURL>.com/api/client/heartbeat?gw_id=CLIENT-0001&amp;sys_uptime=1818&amp;sys_memfree=38676&amp;sys_load=0.02&amp;wifidog_uptime=2">here</a>.</p>
</body></html>

When authenticating it fails and gives the reason: Error: We did not get a valid answer from the central server.

I do have SSL setup in the Wifidog.conf file for the Authserver, but still no joy.

Any suggestions?

benoitg, 2017-10-17:
The gateway does not speak SSL at all, you need to disable whatever https redirect you have configured in Apache for your auth server's virtual host

datashaman, 2017-11-27:
I am running into the same issue. So despite having configuration for SSL in /etc/wifidog.conf, it does not support it? :scream:

datashaman, 2017-11-27:
So I'm guessing the SSL config is just for redirecting the browser to the correct endpoint, but that the gateway to gateway communication does not use HTTPS. Correct?

datashaman, 2017-11-27:
Ok, the following seems to work for me:

```
server {
   listen 80;

   if ($http_user_agent !~ WiFiDog) {
        return 301 https://$host$request_uri;
   }

   # Remember to add config here for wifidog to use.
   # All other browsers will go to the secure site.
}

server {
    listen 443;
    # ...
}
```

datashaman, 2017-11-27:
Apache is *probably* something like this:

```
RewriteEngine On
RewriteCond %{HTTPS} off
RewriteConf %{HTTP_USER_AGENT}  ^WiFiDog
RewriteRule (.*) https://%{HTTP_HOST}%{REQUEST_URI}
```

benoitg, 2017-11-27:
Yes, that would work.

closed 2017-11-27 by benoitg

rakesh2215, 2019-12-05:
Awesome. Many thanks to all contributors.