wifidog/wifidog-gateway #287: Blocking HTTPs traffic

rnicolaev, 2018-01-11:
Is there any way to block incoming HTTPs traffic except Facebook, because we are using facebook login button in our captive portal. But the problem is that we can't specify entire list of fb domains with one command like (www.facebook.com*). And if you will not allow all domains for Facebook the login page display incorrectly. 

Your advise will be highly appreciated. As far as I know this can be easily done using radius server because radius server uses DNS from the router. But unfortunately we are not using radius server.

rnicolaev, 2018-01-11:
Actually my issue is exactly the same as in this post 
https://github.com/wifidog/wifidog-gateway/issues/282

but what as I see, that guy also didn't get an answer, how it can be done in a proper way

benoitg, 2018-01-11:
The short answer is that as the web moves away from http towards systematic use of https, and content delivery networks (CDN) this is becoming increasingly hard to do, and there is an increasing amount of complex code to write.

Basically one needs many additional features for this to work seamlessly:

- A solution to redirect if the first request is https (there are a few, none of them especially good, many quite complex)
- A solution to extract all the domains referenced in a set of webpages (pretty trivial in a web language, pretty nontrivial to do reliably directly on an embedded system).
- A solution to open, AND MAINTAIN AS THEY CHANGE, the whitelisting of all IP aliases for each domain name extracted  above (typically there is more than one IP per domain).
- Even the above will not be compatible with some load balancing solutions, but should work in most case.
- As there are more and more CDN used, one will have to be increasingly carefull to avoid whitelisting the whole of facebook, the whole of google.  Or works, partially enable it enough that if you go to google or facebook, the portal doesn'T splash the login, but not all images or script load.

rnicolaev, 2018-01-11:
Thanks a lot Benoit for your quick reply.
One more question, is there any advantage of using LEDE + WiFidog solution in front of Radius server, in order to prevent user from accessing Internet. As I know radius server uses Java and its quite old.

What is your personal opinion?

benoitg, 2018-01-11:
It really depends on WHY you are doing it.  

However, if for some reason you will have a RADIUS server anyway, you may be better served by a radius-centric solution like https://github.com/coova/coova-chilli

rnicolaev, 2018-01-11:
Our primary aim is to prevent user from accessing internet until he passes
verification process.
On login we show captive portal with login buttons and after we show some
marketing features, like survey, or sponsored ads and etc.
But we noticed that sometimes combination of Lede and Wifidog not very
stable within big number of connections. Our competitors are mainly using
Radius server for the same reason but we don’t like that it purely work on
Java.