wifidog/wifidog-gateway #288: Domain permissions

rnicolaev, 2018-01-18:
Is there any chance to whitelist all URLs from one domain with one command in WiFidog?
For example: 
if i want to whitelist www.facebook.com/profile , www.facebook.com/users can I write something like www.facebook.com/* and all URLs from this domain will be whitelisted.

It will be very useful.

benoitg, 2018-01-19:
Wifidog, or iptables, can neither block or allow specific URLs, only IP adresses.  If you whitelist the ip for www.facebook.com, both www.facebook.com/profile, www.facebook.com/users and all other paths are allowed.   The problem is that there can be more than ont IP per domain name.

See my comment in #287, you basically need all the features I listed except the https redirect.   So:

1. A solution to extract all the domains referenced in a set of webpages (pretty trivial in a web language, pretty nontrivial to do reliably directly on an embedded system).
2. A solution to open, AND MAINTAIN AS THEY CHANGE, the whitelisting of all IP aliases for each domain name extracted above (typically there is more than one IP per domain).
3. Even the above will not be compatible with some load balancing solutions, but should work in most case.
4. As there are more and more CDN used, one will have to be increasingly careful to avoid white-listing the whole of facebook, the whole of google. Or works, partially enable it enough that if you go to google or facebook, the portal doesn't splash the login, but not all images or script load.

mhaas, 2018-01-19:
Wifidog can Whitelist ipsets. If you have your dnsmasq populate the ipset with iOS for Facebook as they are looked up by the client, you should be good.

Check the readme on my fbwlan repo. That should have details.

mhaas, 2018-01-30:
Link here: https://github.com/mhaas/fbwlan/blob/master/README.md#allowing-access-to-facebook

closed 2018-01-30 by mhaas