wifidog/wifidog-gateway #293: SSL: error state on socket

timafh, 2018-03-11:
Hi,

we are using wifidog with ssl on a LEDE device with a self-compiled image.
For our Auth-Server we use a letsencrypt certificate.
The Auth-Server we use is a self-developed c-daemon.

Now, when wifidog is pinging our auth-server over https we every time get cyassl/wolfssl error "error state on socket".

Any ideas what we can do?

benoitg, 2018-03-11:
The gateway does not speak https (and doesn't need to).  Make sure your auth server can speak to the gateway over http.  You should however speak to the client browser only over https.

mhaas, 2018-03-12:
@benoitg I may be confused here, but I think I added support for encrypted auth server communication about three years ago.

@timafh please post logs.

benoitg, 2018-03-12:
@mhaas Humm, so you did.  Sorry about that.   Out of curiosity, what was the rationale behind supporting SSL for gateway to auth communication?

mhaas, 2018-03-13:
I am not sure. Encryption is better than no encryption, and I wouldn't want any Mac addresses etc flying around unencrypted.

I believe the real reason was an implentation detail on my side: Facebook requires my auth server to be reachable via https. And the url would be different between https and non-https as I used a reverse proxy given by my provider.

Still doesn't make a lot of sense. I guess I did it because it felt like the right thing to do :)