wifidog/wifidog-gateway #294: wifidog-ng

zhaojh329, 2018-03-12:
I launched a new project - [wifidog-ng](https://github.com/zhaojh329/wifidog-ng), which no longer uses iptables, but instead writes kernel modules.

Wifidog-ng has been merged into the official Openwrt feeds.

I am here to invite friends interested in wifidog-ng to join.

JJEU, 2018-04-24:
hello,

performance wise, how is wifidog-ng , compared to say, appfree wifidog ?

the performance of appfree is very good, but it only compiles if a special version of SSL is installed. this version is only provided by the team of appfree.
since SSL is a key security, it seems suspicious that a chinese team would want to tinker with it. i am afraid there is a backdoor inside.

still, their performance is very good.
can you test wifidog-ng with the same commands and share the results ?

https://github.com/liudf0716/apfree_wifidog

liudf0716, 2018-12-08:
@JJEU I think u misunderstand our apfree wifidog(not appfree_wifiodg :p), apfree wifidog totally open source, apfree need libevent version > 2.1.8, cause openwrt didn't support that version in current release version but will be used in next release(apfree wifidog also provided PR in openwrt package and approved), for that reason, we port libevent2.1.8 in our sdk environment(base on lede 17.0.1 and open source too). 
since SSL is a key security, we now generated ssl cert by shell automatically instead of using Pre-prepared cert(u can find that in our wifidogx.init file), In fact, even use our Pre-prepared cert, there also no backdoor inside :p 

the latest apfree wifidog totally rewrited by replace libhttpd with libevent, it will become more stable and more efficient.

our openwrt package pr is : 
https://github.com/openwrt/packages/pull/6894
our opensource lede sdk is:
https://github.com/KunTengRom/kunteng-lede-17.01.4