wifidog/wifidog-gateway #295: wifidog iptablesV1.4.10 REDIRECT does not work

linsun78, 2018-03-22:
wifidog log :
(fw_iptables.c:121) iptables command failed(1): iptables -t nat -A WD_br-lan_Unknown -p tcp --dport 80 -j REDIRECT --to-ports 2060
iptables v1.4.10: unknown option `--to-ports'

kernel version:Linux OpenWrt 3.3.8 
iptables version: iptables v1.4.10

benoitg, 2018-03-22:
Where did you get your wifidog package?  It seems you are missing a kernel module (should be a dependency), iptables-mod-nat-extra or some such...

linsun78, 2018-03-23:
I got package on https://github.com/wifidog/wifidog-gateway/ 
how and where can I get iptables-mod-nat-extra module or other module, I think my kernel verison is too low to dependent REDIRECT option