wifidog/wifidog-gateway #296: Wifidog don't block traffic

medrockstar, 2018-04-18:
Hi,

My wifidog don't block traffic

[7][Wed Apr 18 10:33:23 2018][3752](ping_thread.c:198) Auth Server Says: Pong
[7][Wed Apr 18 10:34:21 2018][3752](auth.c:79) Running fw_counter()
[7][Wed Apr 18 10:34:22 2018][3752](gateway.c:245) Handler for SIGCHLD called. Trying to reap a child
[7][Wed Apr 18 10:34:22 2018][3752](gateway.c:249) Handler for SIGCHLD reaped child PID 4014
[7][Wed Apr 18 10:34:22 2018][3752](gateway.c:245) Handler for SIGCHLD called. Trying to reap a child
[7][Wed Apr 18 10:34:22 2018][3752](gateway.c:249) Handler for SIGCHLD reaped child PID 4016
[7][Wed Apr 18 10:34:22 2018][3752](firewall.c:268) Locking client list
[7][Wed Apr 18 10:34:22 2018][3752](firewall.c:268) Client list locked
[7][Wed Apr 18 10:34:22 2018][3752](firewall.c:276) Unlocking client list
[7][Wed Apr 18 10:34:22 2018][3752](firewall.c:276) Client list unlocked
[7][Wed Apr 18 10:34:22 2018][3752](ping_thread.c:74) Running ping()
[7][Wed Apr 18 10:34:22 2018][3752](ping_thread.c:108) Entering ping()
[7][Wed Apr 18 10:34:22 2018][3752](centralserver.c:139) Locking config
[7][Wed Apr 18 10:34:22 2018][3752](centralserver.c:139) Config locked
[7][Wed Apr 18 10:34:22 2018][3752](centralserver.c:189) Level 1: Calculated 1 auth servers in list
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:205) Level 1: Resolving auth server [auth.ilesansfil.org]
[7][Wed Apr 18 10:34:23 2018][3752](util.c:142) Locking wd_gethostbyname()
[7][Wed Apr 18 10:34:23 2018][3752](util.c:142) wd_gethostbyname() locked
[7][Wed Apr 18 10:34:23 2018][3752](util.c:155) Unlocking wd_gethostbyname()
[7][Wed Apr 18 10:34:23 2018][3752](util.c:155) wd_gethostbyname() unlocked
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:264) Level 1: Resolving auth server [auth.ilesansfil.org] succeeded = [108.163.136.106]
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:302) Level 1: Connecting to auth server auth.ilesansfil.org:80
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:331) Level 1: Successfully connected to auth server auth.ilesansfil.org:80
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:141) Unlocking config
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:141) Config unlocked
[7][Wed Apr 18 10:34:23 2018][3752](centralserver.c:147) Connected to auth server
[7][Wed Apr 18 10:34:23 2018][3752](simple_http.c:77) Sending HTTP request to auth server: [GET /ping/?gw_id=90F6522BFAC0&sys_uptime=45772&sys_memfree=6260&sys_load=0.00&wifidog_uptime=603 HTTP/1.0
User-Agent: WiFiDog 1.2.1
Host: auth.ilesansfil.org

]

[7][Wed Apr 18 10:34:23 2018][3752](simple_http.c:87) Reading response
[7][Wed Apr 18 10:34:23 2018][3752](simple_http.c:111) Read 253 bytes
[7][Wed Apr 18 10:34:23 2018][3752](simple_http.c:124) HTTP Response from Server: [HTTP/1.1 200 OK
Date: Wed, 18 Apr 2018 10:34:23 GMT
Server: Apache
X-Powered-By: PHP/5.4.45-0+deb7u11
Set-Cookie: authpuppy=74gc4a6avkdeauugd6n9kost97; path=/
Vary: Accept-Encoding
Connection: close
Content-Type: text/html; charset=utf-8

Pong
]
[7][Wed Apr 18 10:34:23 2018][3752](ping_thread.c:198) Auth Server Says: Pong

root@OpenWrt:~# /etc/init.d/wifidog status
WiFiDog status

Version: 1.2.1
Uptime: 0d 0h 0m 4s
Has been restarted: no
Internet Connectivity: yes
Auth server reachable: yes
Clients served this session: 0

0 clients connected.

Authentication servers:
  Host: auth.ilesansfil.org (108.163.136.106)

benoitg, 2018-04-19:
Are you sure you have GatewayInterface set to the proper interface, your client's side of the router?  Are you sure you are NATing or routing through your router, not just using it as an access point.

medrockstar, 2018-04-19:
My access point is behind router.

how to check NATing or routing ?

my config : 

# without the : separators

# GatewayID default

# Parameter: ExternalInterface
# Default: NONE
# Optional
#
# Set this to the external interface (the one going out to the Inernet or your larger LAN).
# Typically vlan1 for OpenWrt, and eth0 or ppp0 otherwise,
# Normally autodetected

ExternalInterface wlan0

# Parameter: GatewayInterface
# Default: NONE
# Mandatory
#
# Set this to the internal interface (typically your wifi interface).
# Typically br-lan for Openwrt (by default the wifi interface is bridged with wired lan in openwrt)
# and eth1, wlan0, ath0, etc. otherwise
# You can get this interface with the ifconfig command and finding your wifi interface

GatewayInterface br-lan

# Parameter: GatewayAddress
# Default: Find it from GatewayInterface
# Optional
#
# Set this to the internal IP address of the gateway.  Not normally required.

# GatewayAddress 192.168.1.1

# Parameter: HtmlMessageFile
# Default: wifidog-msg.html
# Optional

JJEU, 2018-04-24:
Hello,

there is a problem with your config.
benoit is right, your router is not configurer like a router, it's just an access point.

to have router configuration, you need to edit /etc/config/network
or you can do this via the web interface but it's more difficult.

A definition of a router is a network device with 2 interface : LAN (toward the clients) and WAN (towards the internet).
a router is able to give communication to xxx number of client, accross a single IP on it's WAN interface. this is called NAT.
each client is identified with a single IP adress. this adress is used by the router to distribute the packets to the clients. it's called routing

let's see you /etc/config/network :smiley: config interface 'lan'

config interface **'lan'** _-> this identifies this section as the configuration of the LAN interface. wich name is "lan"_
option ifname **'eth0'** _-> this link the "lan" interface to the hardware interface named "eth0". interfaces are named by the manufacturer like "eth0", "eth1", etc_
option force_link '1'
option type 'bridge' _-> this means that your device just transfers the packet. it's ats like a switch, not like a router. thus NAT is DISABLED and wifidog can't work !_

following is the ip configuration of the lan interface :
option proto 'static'
option netmask '255.255.255.0'
option ip6assign '60'
option ipaddr '192.168.1.66'
option gateway '192.168.1.1'
option dns '192.168.1.1'

----------------------
now lets see your wifidog.conf file

ExternalInterface wlan0 _-> this means that your router is connecting to the internet with the wifi interface. usually it's not the case. in most case you connect to the internet with the RJ45 wire. so there it should be eth1 i guess_

GatewayInterface br-lan _-> this tells your router to use an interface name "br-lan" to connect the clients.
it's better to comment this line so wifidog auto detects, but usually you woud put here the same name as the lan interface configured in /etc/config/network._

-----------------

what is your network looks like ? how it is plugged to the internet ?
also to know what is the name of each physical interface. you would have to check the wiki of openwrt with the name of your device.

Exemple : see here an exemple of a page listing the properties of a random router. if you scroll down you see how the ethernet assignation is done
https://wiki.openwrt.org/toh/pcengines/alix

-----------------------

My advice : 
just download and install the latest stable version of openwrt to your hardware.
by default all the config files would be configured correctly.
you just have to connect in ssh, type "pswd to create a new password, reboot, and enter in the web configuration. go in software, update the list , the search for "wifidog, and install it. install nano too

go in ssh and type nano <path to your wifidog.conf file>
and edit wifidog.conf with the name of your authpuppy server
then connect to the wifi with your cellphone, and try to browse an HTTP website (NOT HTTPS)

medrockstar, 2018-06-30:
Tks,

My installation is like this :

Internet - router - switch - Openwrt (access point)

Your configuration is for this installation ? because I already had router in network.