wifidog/wifidog-gateway #300: wifidog: could not create web server : addess already in use

najwa2, 2018-07-19:
While I'm trying to create a captive portal using wifidog
I'm facing a problem after the installation of wifidog ,I don't know how to solve it
The error that shows after starting wifidog :
wifidog: could not create web server : addess already in use
I set for the openwrt router the address 192.168.1.4 and for the gateway 192.168.1.1 while the address of the wifi's router is 192.168.1.1
I need whenever a client is connected to the wifi, an authentification page shows up but it doesn't seem to be working.
If anyone can just detect where the problem is, I'm thankful

benoitg, 2018-08-06:
Looks like you have another server on the same port, or a leftover wifidog process still running.

najwa2, 2018-08-06:
HI benoitg 
yes exactly, there was another instance running si i had to kill it.
Thank you sir

closed 2018-08-07 by benoitg