wifidog/wifidog-gateway #312: No suporte a new iptables version syntax iptables v1.8.7 (legacy)

deivisonmoraes, 2021-12-07:
Hi, I have a version of openwrt (21.02.1) and iptables 1.8.7. In this version iptables in openwrt is compiled without iptables aliases and some wifidog commands are not entering compromising the functioning of the redirect and other functions. Ex: iptables -t filter -A WiFiDog_br-lan_Internet -m conntrack --cstate INVALID -j DROP
iptables v1.8.7 (legacy): unknown option "--cstate"
Try `iptables -h' or 'iptables --help' for more information.

I would like to know if any of you have already had to work around this situation or if there is any workaround in progress for this version of openwrt?

Thanks!

deivisonmoraes, 2021-12-07:
[image]

henok400, 2022-02-08:
I got the same issue wifidog run work only the first device connection after that other device connect iptable reject and totally wifidog not functions. 
my wifidog conf are 
```
GatewayID F85E3C04D64C
ExternalInterface eth0
GatewayInterface br-lan
AuthServer {
Hostname wan.cloudwifizone.com
HTTPPort 80
SSLAvailable yes
Path /
}
HTTPDMaxConn 253
ClientTimeout 10
PopularServers kernel.org,ieee.org,cloudwifizone.com,ask.com
FirewallRuleSet global {
}
FirewallRuleSet validating-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet known-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet unknown-users {
FirewallRule allow udp port 53
FirewallRule allow tcp port 53
FirewallRule allow udp port 67
FirewallRule allow tcp port 67
FirewallRule block udp port 8000
}
FirewallRuleSet locked-users {
FirewallRule block to 0.0.0.0/0
}
```