wifidog/wifidog-gateway #313: wifidog: could not create web server : addess already in use #300

yassin-git, 2021-12-19:
hello every one , 

i got this error when I tried to launch wifidog for the first time " could not create web server : addess already in use ""

henok400, 2022-02-08:
carfully config the following wifidog.conf setting
```
GatewayID F85E3C04D64C
ExternalInterface eth0
GatewayInterface br-lan
AuthServer {
Hostname wan.cloudwifizone.com
HTTPPort 80
SSLAvailable yes
Path /
}
HTTPDMaxConn 253
ClientTimeout 10
PopularServers kernel.org,ieee.org,cloudwifizone.com,ask.com
FirewallRuleSet global {
FirewallRule allow tcp to cloudwifizone.com
FirewallRule allow tcp to checkout.flutterwave.com
FirewallRule allow tcp to assets.flutterwave.com
FirewallRule allow tcp to flutterwave.com
FirewallRule allow tcp to filestackcontent.com
FirewallRule allow tcp to cdn.filestackcontent.com
FirewallRule allow tcp to eligrey.com
FirewallRule allow tcp to purl.eligrey.com
FirewallRule allow tcp to ravepay.co
FirewallRule allow tcp to api.ravepay.co
FirewallRule allow tcp to herokuapp.com
FirewallRule allow tcp to coreflutterwaveprod.com
FirewallRule allow tcp to cardinalcommerce.com
FirewallRule allow tcp to authentication.cardinalcommerce.com
}
FirewallRuleSet validating-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet known-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet unknown-users {
FirewallRule allow udp port 53
FirewallRule allow tcp port 53
FirewallRule allow udp port 67
FirewallRule allow tcp port 67
FirewallRule block udp port 8000
}
FirewallRuleSet locked-users {
FirewallRule block to 0.0.0.0/0
}

```

closed 2022-02-12 by yassin-git

moimenashraf1, 2025-03-07:
Solved
The error occurs because wifidog or another service is already using the same port. you have to stop already working Wifidog service or  You may need to check which process is using the port and adjust the configuration if needed
Try stopping Wifidog ..  just do 
**/etc/init.d/wifidog stop** 
then start it manually with Command
 **wifidog**