wifidog/wifidog-gateway #314: Wifidog error iptables_fw_counters_update(): Could not find 0 in client list

henok400, 2022-04-20:
i trying 2nd device connect wifidog getting this error and connection disconnected
```
Wed Apr 20 19:28:32 2022 daemon.info wifidog[3902]: Got ALLOWED from central server authenticating token pa1sufibywwal371ffuf7xnyz46mf3yb from 192.168.1.205 at 16:4c:8f:82:87:f5 - adding to firewall and redirecting them to portal
Wed Apr 20 19:29:04 2022 daemon.err wifidog[3902]: iptables_fw_counters_update(): Could not find 0 in client list, this should not happen unless if the gateway crashed
Wed Apr 20 19:29:04 2022 daemon.err wifidog[3902]: Preventively deleting firewall rules for 0 in table WiFiDog_$ID$_Outgoing
```
Someone help me how to fix this @acv and

wsch32, 2022-04-24:
I have the same problem in Openwrt 21.02, and I fixed.

The problem is in the function of  iptables_fw_counters_update() ( fw_iptables.c   line 1228)

The reason why this problem occur is as following:
1. fscanf format string problem(last %*s );
2. because of reason 1, the the ip string got a  result of "0",and the C function inet_aton() take the 
   result of "0" for granted (return true).
3. The funcion client_list_find_by_ip() can't find the ip of "0",and trigger the reset of firwall logic.

That's All!

Talk is cheap, show the code(fw_iptables.c   line 1228) .........

`
/** Update the counters of all the clients in the client list */
int
iptables_fw_counters_update(void)
{
	FILE *output;
	char *script, ip[16] = {0}, rc;
	unsigned long long int counter;
	t_client *p1;
	struct in_addr tempaddr;

	/* Look for outgoing traffic */
	safe_asprintf(&script, "%s %s", "iptables", "-v -n -x -t mangle -L " CHAIN_OUTGOING);
	iptables_insert_gateway_id(&script);
	debug(LOG_DEBUG, "Run iptables Command: %s", script);
	output = popen(script, "r");
	free(script);
	if (!output) {
		debug(LOG_ERR, "popen(): %s", strerror(errno));
		return -1;
	}

	LOCK_CLIENT_LIST();
	reset_client_list();
	UNLOCK_CLIENT_LIST();

	/* skip the first two lines */
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (('\n' != fgetc(output)) && !feof(output)) ;

	while (output && !(feof(output))) {
		rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %15[0-9.] %*s %*s %*s %*s %*s", &counter, ip);
		//rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %15[0-9.] %*s %*s %*s %*s %*s 0x%*u", &counter, ip);
		debug(LOG_DEBUG, "Read from iptables format string output: ip = %s  count = %llu", ip, counter);
		if (2 == rc && EOF != rc) {
			/* Sanity */
			if ( (!inet_aton(ip, &tempaddr)) || (0 == strcmp(ip,"0")) ) {
				debug(LOG_WARNING, "I was supposed to read an IP address but instead got [%s] - ignoring it", ip);
				continue;
			}
			debug(LOG_DEBUG, "Read outgoing traffic for %s: Bytes=%llu", ip, counter);
			LOCK_CLIENT_LIST();
			if ((p1 = client_list_find_by_ip(ip))) {
				if ((p1->counters.outgoing - p1->counters.outgoing_history) < counter) {
					p1->counters.outgoing_delta = p1->counters.outgoing_history + counter - p1->counters.outgoing;
					p1->counters.outgoing = p1->counters.outgoing_history + counter;
					p1->counters.last_updated = time(NULL);
					debug(LOG_DEBUG, "%s - Outgoing traffic %llu bytes, updated counter.outgoing to %llu bytes.  Updated last_updated to %d", ip,
						  counter, p1->counters.outgoing, p1->counters.last_updated);
					p1->is_online = 1;
				}

				// get client name
				if(p1->name == NULL)
					__get_client_name(p1);

				if(p1->wired == -1) {
					p1->wired = br_is_device_wired(p1->mac);
				}
				UNLOCK_CLIENT_LIST();
			} else {
				UNLOCK_CLIENT_LIST();
				debug(LOG_ERR,
					  "iptables_fw_counters_update(): Could not find %s in client list, this should not happen unless if the gateway crashed",
					  ip);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_OUTGOING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_OUTGOING, ip, NULL, 5);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_INCOMING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_INCOMING, ip, NULL, 5);
			}

		}
	}
	pclose(output);

	/* Look for incoming traffic */
	safe_asprintf(&script, "%s %s", "iptables", "-v -n -x -t mangle -L " CHAIN_INCOMING);
	iptables_insert_gateway_id(&script);
	debug(LOG_DEBUG, "Run iptables Command: %s", script);
	output = popen(script, "r");
	free(script);
	if (!output) {
		debug(LOG_ERR, "popen(): %s", strerror(errno));
		return -1;
	}

	/* skip the first two lines */
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (output && !(feof(output))) {
		rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %*s %15[0-9.]", &counter, ip);
		if (2 == rc && EOF != rc) {
			/* Sanity */
			if ( (!inet_aton(ip, &tempaddr)) || (0 == strcmp(ip,"0")) ) {
				debug(LOG_WARNING, "I was supposed to read an IP address but instead got [%s] - ignoring it", ip);
				continue;
			}
			debug(LOG_DEBUG, "Read incoming traffic for %s: Bytes=%llu", ip, counter);
			LOCK_CLIENT_LIST();
			if ((p1 = client_list_find_by_ip(ip))) {
				if ((p1->counters.incoming - p1->counters.incoming_history) < counter) {
					p1->counters.incoming_delta = p1->counters.incoming_history + counter - p1->counters.incoming;
					p1->counters.incoming = p1->counters.incoming_history + counter;
					debug(LOG_DEBUG, "%s - Incoming traffic %llu bytes, Updated counter.incoming to %llu bytes", ip, counter, p1->counters.incoming);
				/*	p1->counters.last_updated = time(NULL); */
				}

				UNLOCK_CLIENT_LIST();
			} else {
				UNLOCK_CLIENT_LIST();
				debug(LOG_ERR,
					  "iptables_fw_counters_update(): Could not find %s in client list, this should not happen unless if the gateway crashed",
					  ip);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_OUTGOING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_OUTGOING, ip, NULL, 5);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_INCOMING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_INCOMING, ip, NULL, 5);
			}
		}
	}
	pclose(output);

	return 1;
}
`

wsch32, 2022-04-24:
> i trying 2nd device connect wifidog getting this error and connection disconnected
>
> ```
> Wed Apr 20 19:28:32 2022 daemon.info wifidog[3902]: Got ALLOWED from central server authenticating token [...]

> i trying 2nd device connect wifidog getting this error and connection disconnected
>
> ```
> Wed Apr 20 19:28:32 2022 daemon.info wifidog[3902]: Got ALLOWED from central server authenticating token [...]

I have the same problem in Openwrt 21.02, and I fixed.

The problem is in the function of  iptables_fw_counters_update() ( fw_iptables.c   line 1228)

The reason why this problem occur is as following:
1. fscanf format string problem(last %*s );
2. because of reason 1, the the ip string got a  result of "0",and the C function inet_aton() take the 
   result of "0" for granted (return true).
3. The funcion client_list_find_by_ip() can't find the ip of "0",and trigger the reset of firwall logic.

That's All!

Talk is cheap, show the code(fw_iptables.c   line 1228) .........

`
/** Update the counters of all the clients in the client list */
int
iptables_fw_counters_update(void)
{
	FILE *output;
	char *script, ip[16] = {0}, rc;
	unsigned long long int counter;
	t_client *p1;
	struct in_addr tempaddr;

	/* Look for outgoing traffic */
	safe_asprintf(&script, "%s %s", "iptables", "-v -n -x -t mangle -L " CHAIN_OUTGOING);
	iptables_insert_gateway_id(&script);
	debug(LOG_DEBUG, "Run iptables Command: %s", script);
	output = popen(script, "r");
	free(script);
	if (!output) {
		debug(LOG_ERR, "popen(): %s", strerror(errno));
		return -1;
	}

	LOCK_CLIENT_LIST();
	reset_client_list();
	UNLOCK_CLIENT_LIST();

	/* skip the first two lines */
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (('\n' != fgetc(output)) && !feof(output)) ;

	while (output && !(feof(output))) {
		rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %15[0-9.] %*s %*s %*s %*s %*s", &counter, ip);
		//rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %15[0-9.] %*s %*s %*s %*s %*s 0x%*u", &counter, ip);
		debug(LOG_DEBUG, "Read from iptables format string output: ip = %s  count = %llu", ip, counter);
		if (2 == rc && EOF != rc) {
			/* Sanity */
			if ( (!inet_aton(ip, &tempaddr)) || (0 == strcmp(ip,"0")) ) {
				debug(LOG_WARNING, "I was supposed to read an IP address but instead got [%s] - ignoring it", ip);
				continue;
			}
			debug(LOG_DEBUG, "Read outgoing traffic for %s: Bytes=%llu", ip, counter);
			LOCK_CLIENT_LIST();
			if ((p1 = client_list_find_by_ip(ip))) {
				if ((p1->counters.outgoing - p1->counters.outgoing_history) < counter) {
					p1->counters.outgoing_delta = p1->counters.outgoing_history + counter - p1->counters.outgoing;
					p1->counters.outgoing = p1->counters.outgoing_history + counter;
					p1->counters.last_updated = time(NULL);
					debug(LOG_DEBUG, "%s - Outgoing traffic %llu bytes, updated counter.outgoing to %llu bytes.  Updated last_updated to %d", ip,
						  counter, p1->counters.outgoing, p1->counters.last_updated);
					p1->is_online = 1;
				}

				// get client name
				if(p1->name == NULL)
					__get_client_name(p1);

				if(p1->wired == -1) {
					p1->wired = br_is_device_wired(p1->mac);
				}
				UNLOCK_CLIENT_LIST();
			} else {
				UNLOCK_CLIENT_LIST();
				debug(LOG_ERR,
					  "iptables_fw_counters_update(): Could not find %s in client list, this should not happen unless if the gateway crashed",
					  ip);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_OUTGOING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_OUTGOING, ip, NULL, 5);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_INCOMING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_INCOMING, ip, NULL, 5);
			}

		}
	}
	pclose(output);

	/* Look for incoming traffic */
	safe_asprintf(&script, "%s %s", "iptables", "-v -n -x -t mangle -L " CHAIN_INCOMING);
	iptables_insert_gateway_id(&script);
	debug(LOG_DEBUG, "Run iptables Command: %s", script);
	output = popen(script, "r");
	free(script);
	if (!output) {
		debug(LOG_ERR, "popen(): %s", strerror(errno));
		return -1;
	}

	/* skip the first two lines */
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (('\n' != fgetc(output)) && !feof(output)) ;
	while (output && !(feof(output))) {
		rc = fscanf(output, "%*s %llu %*s %*s %*s %*s %*s %*s %15[0-9.]", &counter, ip);
		if (2 == rc && EOF != rc) {
			/* Sanity */
			if ( (!inet_aton(ip, &tempaddr)) || (0 == strcmp(ip,"0")) ) {
				debug(LOG_WARNING, "I was supposed to read an IP address but instead got [%s] - ignoring it", ip);
				continue;
			}
			debug(LOG_DEBUG, "Read incoming traffic for %s: Bytes=%llu", ip, counter);
			LOCK_CLIENT_LIST();
			if ((p1 = client_list_find_by_ip(ip))) {
				if ((p1->counters.incoming - p1->counters.incoming_history) < counter) {
					p1->counters.incoming_delta = p1->counters.incoming_history + counter - p1->counters.incoming;
					p1->counters.incoming = p1->counters.incoming_history + counter;
					debug(LOG_DEBUG, "%s - Incoming traffic %llu bytes, Updated counter.incoming to %llu bytes", ip, counter, p1->counters.incoming);
				/*	p1->counters.last_updated = time(NULL); */
				}

				UNLOCK_CLIENT_LIST();
			} else {
				UNLOCK_CLIENT_LIST();
				debug(LOG_ERR,
					  "iptables_fw_counters_update(): Could not find %s in client list, this should not happen unless if the gateway crashed",
					  ip);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_OUTGOING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_OUTGOING, ip, NULL, 5);
				// debug(LOG_ERR, "Preventively deleting firewall rules for %s in table %s", ip, CHAIN_INCOMING);
				// __iptables_fw_destroy_mention("mangle", CHAIN_INCOMING, ip, NULL, 5);
			}
		}
	}
	pclose(output);

	return 1;
}
`

henok400, 2022-04-25:
Where can i run this script in openwrt device?

wsch32, 2022-04-25:
> Where can i run this script in openwrt device?

you should fix the sourse code as my advice and rebuild it

henok400, 2022-04-25:
Could you upgrade wifidog this some script suggestion @aparcar @lipnitsk @jefferyto

henok400, 2022-04-26:
@wsch32 Could direct me how to compile wifidog-gateway or suguest link to setup .ipk Thank you!

wsch32, 2022-04-29:
> @wsch32 Could direct me how to compile wifidog-gateway or suguest link to setup .ipk Thank you!

what kind of device(device model or cpu) do you want to install the wifidog?leave me a email address,I'll send the ipk file to you.

henok400, 2022-04-29:
I have 100 type of model i use but now Gl.inet 6416 and Mt300n mail me berlikplc@gmail.com i collect many how to compile bit.do/domain100

predators46, 2023-02-16:
@wsch32 

can you make a patch for this